physee/Victron.Arduino-ESP8266 #5: Values not updating

DPOdendaal, 2019-05-06:
I am trying to use this code to send data to Blynk as an output rather than serial.
The values seem to not update throughout most of the day. 
Any ideas?
Thanks

deswong, 2020-04-07:
Have you checked to see if this is because of the checksum value being incorrect? I have noticed on my test bench that the checksum sometimes is not correct and this is why I have had issues with my values updating. A better and shielded connector resolves this problem for me.

padigree, 2020-08-28:
Same for me. Keeping the values for about 15 measurements, independent of any timer or whatever.. After it stops and only keep working again if I reset my esp. Wouldn't it be possible to simply delete/ignore  the wrong data if the checksum isn't correct? I'm not that great in programming, so just an opinion.

padigree, 2020-08-28:
Short update: neither with another cable, nor with another esp or different timings it does work. It's not the checksum, seems like the communication freezes and after ~2 minutes, you'll always receive the exactly same data. Tried in blynk and tried via serial :( any suggestions?

metrafonic, 2023-02-27:
For me the config.h fields were not correct for the latest victron update, so my checksum was never valid. I ended up rewiting most of the data retrieval, which i packaged in a library here https://github.com/metrafonic/arduino-vedirect

Booli, 2024-10-23:
Hey @metrafonic,

would have greatly appreciated a pull request with new config.h settings for different firmware. 

A note on your new implementation, it uses the [.readStringUntil()](https://github.com/metrafonic/arduino-vedirect/blob/main/src/VEDirect.cpp#L28) command, which is a blocking command. This library was purposely written to not use any of the blocking serial read commands, as it will mess with other loops on the ESP. For example TCP/IP / MQTT etc.

closed 2024-10-28 by Booli