physee/Victron.Arduino-ESP8266 #10: get Day Sequence Values

gsus24, 2021-03-16:
Thanks for sharing. Works great and same results as with usb serial.
But how to get the values from each day? 

Like min/max battery voltage, charge state yield and so on, like the values in the official app shown.

I monitored serial usb connection when connecting to the app, but can´t find out how to receive this values.
Hope you have some tips for me.

gsus24, 2021-03-16:
I answer myself
Its well documented in paper "VE.Direct HEX Protocol MPPT" 
Can be found on the bottom of the victron site
[https://www.victronenergy.com/support-and-downloads/technical-information](url)
Look like lot of work.

If somebody have some useful links or other projects, please let me know

Booli, 2021-04-01:
Hey @gsus24, this only uses the standard 1sec output of the MPPT. It does not support the HEX protocol. So if you want that type of information, a new library needs to be written.

closed 2021-04-01 by Booli