physee/Victron.Arduino-ESP8266 #12: MPPT 75/15 checksum always wrong

muelli01, 2021-08-23:
I have a MPPT 75/15 Bluesolar with newest FW 1.56.
Sadly I had to bypass the 
"if (!checksum)" part, because the checksum was never 0, in fact it is always calculated wrong and 3 higher than the MPPTs transmitted checksum.

Example:
If the serially transmitted checksum by the MPPT is 66, the calculated checksum is 69.
If the serially transmitted checksum by the MPPT is 61, the calculated checksum is 64.
etc.

That is quite strange, as there don't seem to be any errors during transmission....those values are not fluctuating.

mike2nl, 2021-08-24:
@muelli01 
Hi, 
-> after locking inmto the source and the 3 higher value could it be in the source at line 123...124 (comment)?
-> could it be that the number of num_keywords is different (mppt_75_10 <> mppt_75_15) so the array get no correct checksum?

Also have you the config for the MPPT_75_15 and can you share it please?
I use the SmartSolar MPPT 75/15 (Bluetooth version).

muelli01, 2021-08-24:
OMG, I am so stupid. You are totally correct, the 75/15 has more values than the 75/10. 
I did not even think that the fields could be different in those models. Will read them asap and then share the updated config here!
Thanks for that hint!

muelli01, 2021-08-24:
75/15 has one more field (MPPT). 
I added that field to the config and now it works. Assumption is the mother of all.... you know :)

Here is the config:

// MPPT 75 | 15
#ifdef MPPT_75_15

	const byte buffsize = 32;
	const byte value_bytes = 33;
	const byte label_bytes = 9;
	const byte num_keywords = 19;

	char keywords[num_keywords][label_bytes] = {
	"PID",
	"FW",
	"SER#",
	"V",
	"I",
	"VPV",
	"PPV",
	"CS",
    "MPPT",
	"ERR",
	"LOAD",
	"IL",
	"H19",
	"H20",
	"H21",
	"H22",
	"H23",
	"HSDS",
	"Checksum"
	};
	#define PID 0
	#define FW 1
	#define SER 2	// Offically SER# but # does not play that well as macro
	#define V 3     // ScV
	#define I 4     // ScI
	#define VPV 5   // PVV
	#define PPV 6   // PVI = PVV / VPV
	#define CS 7    // ScS
       #define MPPT 8
	#define ERR 9   // ScERR
	#define LOAD 10  // SLs
	#define IL 11   // SLI
	#define H19 12
	#define H20 13
	#define H21 14
	#define H22 15
	#define H23 16
	#define HSDS 17
	#define CHECKSUM 18
#endif

closed 2022-10-13 by Booli